abersailbot/dewi-hardware #11: fully water electronics and battery boxes

naturesyouth, 2015-09-04:
improve the water proofing on the internal boxes

Tebazil12, 2016-12-21:
This should be mostly solved by now using cables (instead of lots of individual wires) as the glands tighten nicely around them and should seal. But, a few of the glands look very bent out of shape and a couple are missing the rubber bit that helps seal them, so this need looking into to make sure they are definitely waterproof.

Danyc0, 2016-12-22:
The "rubber bit that helps seal them" is actually the bit that does all the waterproofing, without it they're not waterproof. There should be some glands around the place which can be cannibalised for the seals.

Danyc0, 2016-12-22:
Also, are there any glands on the boxes which don't have any wires in them at all? As they will likely cause an issue. In the past we've filled them with hot glue as a temporary measure, but that's not gonna be IP67!

Tebazil12, 2016-12-22:
Do you know the size/make of the glands so we can find spare parts? Someone suggested we find the top bit of the gland without a hole for wires, so its covered and watertight.

Danyc0, 2016-12-22:
@colinsauze might know. What do you mean by "find the top bit of the gland without a hole for wires, so its covered and watertight" ?

Danyc0, 2016-12-22:
What really needs to happen, is just to do #55 and use some sort of waterproof connecter (similar to that which is in kitty), instead of glands