abersailbot/dewi-hardware #16: Power monitoring

colinsauze, 2015-09-04:
Monitor battery voltage and currents on various devices.

colinsauze, 2016-11-30:
We are currently looking at using the INA219 at work. Its an I2C current/voltage monitor able to monitor up to 3.2A and 26V. See https://www.adafruit.com/product/904

naturesyouth, 2016-11-30:
that looks pretty cool, definitely worth looking into.

naturesyouth, 2016-11-30:
could this be spun into board in the battery box? that way we can monitor both sides of the power system?

Danyc0, 2016-12-01:
IIRC, based on the measurements I took last year, with both wifi dongles running, especially if there's now a Pi3 (plz no), and we're doing a lot of CV we will overrun that 3.2A limit

naturesyouth, 2016-12-01:
I would think we would have one chip per channel / battery rather then one for everything. and was that including servos? because the power supply for the logic isn't rated for anything more then an amp and a half.

Danyc0, 2016-12-01:
Now I come to think of it, I'm not sure :P Please disregard that comment :L

colinsauze, 2016-12-01:
It is possible (but hard, needs surface mount soldering) to change the shunt resistor and adjust its measurement range. But 3.2A max for the whole boat sounds about right to me. If there are multiple sensors then you shouldn't saturate them.

TomGC96, 2016-12-01:
Yeah you do not remember correctly, the measurements for current I took for the Pi, Arduino and Captain Morgan hardly reached 0.5A.