abersailbot/dewi-hardware #19: Shield all cables that go outside the box.

colinsauze, 2015-09-04:
Use shielded cables whereever possible on signal lines. Twist power and ground cables together.

Tebazil12, 2016-12-21:
The data wire (yellow) leading to the rudder servo is now shielded. Only the wire leading to the sailwinch should now need shielding.

Danyc0, 2016-12-22:
Just to double check, is one (and only one) end of all the shielding attached to an earth?

Danyc0, 2016-12-22:
And are the ground and power lines twisted?

Tebazil12, 2016-12-22:
Not earthed yet, most of the wires haven't been reattached to the bits inside the boxes after I replaced the wires with cable. And no, the power and ground are not twisted, should they be?

Danyc0, 2016-12-22:
Ah ok! It's in the note for this issue to twist them together, it's just another thing to try and combat interference

Tebazil12, 2016-12-22:
I didn't see that bit, whoops. The power lines are mostly cable now, so theres no way of twisting them together.

Danyc0, 2016-12-22:
As long as they're close enough, it should be ok

Tebazil12, 2017-05-24:
ok, so at this stage only the data wire to the rudder has been shielded, not the sail winch. I am not sure if it is worth redoing the wiring again to make it shielded (unless interference really is a big problem)

Danyc0, 2017-05-24:
Interference, specifically on the sail winch has been a very big problem in the past. Especially with certain RC remotes. When interference happens on the winch, it can often go past it's end stops and try to draw the sail in too far, which then breaks the winch. It's probably the most important cable of them all. Definitely worth doing, although if it's fine for now, then it should be ok for testing, but competitions have a lot more RF interference from everyone else's remotes, which is where the noise was worst, so it should probably be done before Sailbot.