abersailbot/dewi-hardware #30: Better handles.

colinsauze, 2015-09-04:
Try to improve handles so they are more comfortable and larger. Front handle is structural for the jib to deck connection.

TomGC96, 2016-06-17:
Since the front handle has broken off it will need to be fully replaced or removed. Currently the front block has been screwed to the deck and an eyelet has been attached to connect the jib boom.

TomGC96, 2016-09-12:
The front handle has now been fully replaced with a stainless steel eye plate screwed in with stainless steel screws and sealed with silicone. This allows the jib to be tied down, as well as allowing us to use a rope handle which can be detached when necessary. It has been tested and found to be working satisfactorily, though the rope handle could be made a little more comfortable by surrounding it with PVC tubing. This closes the issue for the time being, but it can be re-opened if it is found to be unsuitable.

closed 2016-09-12 by TomGC96