abersailbot/dewi-hardware #48: Alternative methods of Arduino programing

naturesyouth, 2016-02-14:
It would be nice if someone could look into the possibility of programming the Arduino without a usb connection, either over spi or serial, preferably with some hardware experimentation. This is in refrence to #45

Danyc0, 2016-02-14:
So this page explains how to program an arduino using another arduino:
https://www.arduino.cc/en/Tutorial/ArduinoISP
I reckon if you reimplemented the ArduinoISP example (found built into the IDE these days) for the Pi it might just be possible. However I think that would take too much time to be worth it just for the purpose of getting rid of a USB cable

naturesyouth, 2016-12-02:
This issue was moved to abersailbot/captain-h-morgan#4

closed 2016-12-02 by naturesyouth