abersailbot/dewi-hardware #53: Re-paint Dewi

TomGC96, 2016-06-18:
The entire boat needs re-painting. Paint has started flaking off in places and is causing the wood to absorb some water. The bottom of the keel where a repair has been made should be re-painted sooner as it stands out.

kragniz, 2016-06-18:
More importantly, the balsa around the edge needs to be resealed. This should probably be done with something better than paint, then painted over when that is done.

TomGC96, 2017-05-27:
The balsa around the edge has been sealed with 2 coats of epoxy, and is ready to be re-painted.

TomGC96, 2017-06-01:
All holes in the hull of the boat have been filled in, and the entire boat has been sanded smooth in order to remove paint. We have also primed the hull with 2 coats of PlastiKote Grey Primer spray - she is now ready to be painted to her famous colours, with PlastiKote Gloss Super spray paint.

TomGC96, 2017-06-02:
The deck and top edges of the hull have now been painted yellow. This will be dry by Monday. I will then paint the orange hull of the boat on Tuesday night, in order for it to dry by the time we depart on Friday.

Danyc0, 2017-06-02:
What's the plan for the orange painting? Straight line or the nice wavy thing we had recently? And if the latter, how do you plan of doing the wavy bit?

TomGC96, 2017-06-06:
To be honest, I hadn't realised it was on purpose until I was told, I thought it was a straight line gone wrong. How do people want me to do it?

kragniz, 2017-06-06:
I personally like wavy

TomGC96, 2017-06-06:
How did you do the waviness? Or was it just guesswork? That could be difficult to get right.

Tebazil12, 2017-06-06:
Looking at the old paint, they had a straight line then made it curved after. You could put a straight line of masking tape all the way around, then put more tape on at the front and back to make the curve, if that makes sense? (There might be an easier way though)

TomGC96, 2017-06-08:
A different paint was used for the underside of the boat in the end, which was Rust-Oleum Quick Colour Orange. Dewi has now been fully re-painted to her original colours!

closed 2017-06-08 by TomGC96