abersailbot/dewi-hardware #57: More robust way to connect wires to the Arduino

NicholasRostant, 2016-09-05:
The current method is not particularly reliable and difficult to notice when a connection has gone wrong.

naturesyouth, 2016-09-05:
in the kit i sent with the new gps board there a collection of 2.54mm right angle latching connectors, can these be soldered to the underside of arduino pins? That would make it almost impossible to plug the wire in the wrong way, and be a lot more robust than the current connection.

Tebazil12, 2016-11-30:
Currently changing from an uno to a nano. Plan so far is to use veriboard and headerpins to connect to the nano instead of just headerpins, and attach everything to the lid of the box, instead of the bottom. This should stop the wires being tugged on every time the box is opened.

Danyc0, 2016-12-01:
What's the plan for securing it to the lid?

Tebazil12, 2016-12-01:
Currently hot glue, but still trying to think if theres anything better.

Just to clarify, by 'attach everything to the lid of the box' I meant nano and veriboard, not Capt Morgan and pi (those will remain in the bottom of the box).

Tebazil12, 2017-05-24:
this is mostly complete now, the arduino just needs testing to make sure everything was rewired correctly