abersailbot/dewi-hardware #58: Find an alternative switch mode regulator for Captain H Morgan

Danyc0, 2016-09-08:
The DC-DC converter on CHM overheats and starts cutting its output at 71 degrees C, causing brown-outs on the arduino (but not the pi, probably because it can deal with a lower voltage). It then cuts out completely at 85 degrees C IIRC. An alternative is to find a way to keep everything cooler, because the whole board stack gets very (very) hot.

naturesyouth, 2016-09-08:
which rev of the board is this fault on? also can we get a current reading on the batt supply with everything plugged in?

Danyc0, 2016-09-08:
V1.2, and no. I'm pretty sure we're going over the 1A limit, we should use this instead: R-78Bxx-1.5(L), it supports up to 1.5A

naturesyouth, 2016-09-08:
hmm, we should already be using the 1.5A dc-dc, and when we did out calculations we should have been well under an amp...

TomGC96, 2016-09-09:
@naturesyouth is correct, yes, we are using the 1.5A regulator.

Danyc0, 2016-09-13:
Ah ok, well we still need to either find a replacement, or find another way to keep the box cool, as it's very much overheating. There is currently a tiny python script I wrote sitting on the Pi called temp.py which will both log and output the Pi's chip temperature once every 5 seconds

Danyc0, 2016-09-13:
The Arduino also has a cut off at 85 degrees C so maybe we just need to find a way to keep the box cooler. Increasing the height of the standoffs might help

naturesyouth, 2016-11-21:
can we run a test for whats making the most heat, i recon the best way to do this is setup a second arduino with some temp sensors fixed to various parts of the electronics as well as inside and outside the box, and then running  untill the overheat happens,

Danyc0, 2016-11-21:
Might be a good idea, there's a temperature sensor built into the IC on the Pi, and I build a rubbish little script which logs its data, but it only exists on the Pi we took to WRSC

naturesyouth, 2016-12-02:
This issue was moved to abersailbot/captain-h-morgan#3

closed 2016-12-02 by naturesyouth