abersailbot/dewi-hardware #59: Fix power cables in the battery box

Danyc0, 2016-09-08:
The power cables in the battery box (including those coming directly out of the batteries) are very badly fitting and constantly cause reboots of the Pi and the Servos. The cables need replacing and a better solution to the wires coming from the batteries needs to be found

Danyc0, 2017-06-02:
Has anything been done to this? because it was a pretty major problem

TomGC96, 2017-06-02:
It hasn't been done yet, however I think we agreed the best solution would probably be to cut the cables on the batteries and attach Dean's connectors. I intend to do this during the coming week.

Danyc0, 2017-06-02:
A fair solution, just keep in mind it means you won't be able to borrow batteries at the competition, so make sure you take spares!

TomGC96, 2017-06-04:
It wouldn't make any difference in that sense, as we would still take the Tamiya to Dean's converter cable - so switching back would not be an issue.