abersailbot/dewi-hardware #60: New rudder

kragniz, 2016-09-11:
Our current rudder is too short and cannot control the boat properly when it is heeling in strong winds. The current rudder could be extended to match the original modified rudder that was lost, but a clean start would be nicer.

TomGC96, 2017-02-27:
I've contacted Paul Miller to ask about the rudder. Although he mentions that the transom hung rudder is the least efficient, the new rudder should be a NACA 0012 airfoil section, 75mm in chord, and the length should be to the bottom of the keel, -1cm to avoid damaging the rudder when putting the boat down or running her aground (hopefully this won't happen).

TomGC96, 2017-06-01:
After speaking to Barry, it became apparent that milling balsa to form a rudder would be difficult as it will be very thin. It seems a better way to make a replacement rudder, would be to cut various small pieces of NACA 0012 airfoil section, and joining them together as seen here. This should then be sanded, shaped, and sealed.
[image]

Danyc0, 2017-06-01:
Would something like high density foam be better for milling? Because I think that's better than balsa in almost every way re: rudders.

TomGC96, 2017-06-01:
This was also discussed previously with Barry, and the answer was no, as it would likely be more flexible and not as firm. It would also add unnecessary weight, and is expensive. But as mentioned, the suggestion was that avoiding milling would be best for our situation.

Danyc0, 2017-06-01:
My memory of the foam we used for kitty, was it was like 10 pounds for the amount we'd need. I'm pretty sure it's also lighter. Also, all the rigidity comes from the epoxy/fibreglass, the foam is effectively just a mould, ideally it could be dissolved after the fibreglass is added. That's why @kragniz used it for kitty.

colinsauze, 2019-08-21:
Completed by @linemcr

closed 2019-08-21 by colinsauze