abersailbot/dewi-hardware #61: Add a cooling system to the electronics box

TomGC96, 2016-09-13:
In WRSC in Portugal we found the switch mode regulator overheating and causing a loss in power. At the moment we have it mounted on a PCB between the Pi and Arduino. A fan may be enough to create some airflow within the box, or we could be ambitious and create a water cooling system pumping water from outside the boat.

CloudCoppola, 2017-03-08:
Couldn't we just put in a gel cooling pad with the electronics?

CloudCoppola, 2017-05-27:
Best to measure inside the box (width, length and height). But I've found this
https://www.amazon.co.uk/gp/product/B00INCVS08/ref=s9_dcacsd_dcoop_bw_c_x_3_w?th=1

Danyc0, 2017-05-30:
Aren't those things basically just the capitalist version of a tea towel. I think it might get burnt if it was stuck to the pi.
If however you were talking about the cooling pack itself (which that is totally not the link for), I'd say that they're gonna create loads of condensation and moisture in the box and should really be avoided

CloudCoppola, 2017-06-02:
I'm not sure if your on about the ice pack or gel pack which I was mentioning, but It was either that idea or redesigning Dewi with a water cooling system. Better to do that than adding a heat sink and cooling fan for obvious reasons.

naturesyouth, 2017-06-02:
If i recall the problem only occurred when the boat was out of the water in a sealed box with high ambient temperature, so having the box open and a fan would solve the issue, just make sure the boat isn't running sealed up on the dock. no major hardware mods needed.

CloudCoppola, 2017-06-02:
Then that's just as easy as buying a cheap handheld battery powered fan and keeping it inside the box until she goes out again.

Danyc0, 2017-06-02:
The gel packs still cause condensation IIRC. I'd say just a heat-sync on the Pi would be enough, the problem is that it's all so close together with Captain Morgan now, so a good heat-sync which gets the heat away from the super enclosed space between the Pi and the PCB should do the job.