abersailbot/dewi-hardware #67: Remove brass plate

TomGC96, 2016-12-01:
The brass plate reinforcing the rudder pin collar has been affected by water resulting in copper oxide ending up on the deck. 2 screws have also rusted due to them not being stainless steel. These should be removed and replaced.

CloudCoppola, 2016-12-14:
Do we have a spare brass plate somewhere?

TomGC96, 2016-12-14:
If we use brass again the same will happen instantly. It's better to use something else.

naturesyouth, 2016-12-14:
you should switch to an aluminum plate, if you can't find one lying around you can one off ebay for cheap

TomGC96, 2016-12-14:
Aluminium also corrodes in salt water, although a little less! I'm not sure the plate is the way to go. It might be worth replacing the collar in some way.

CloudCoppola, 2016-12-14:
Well brass is waterproof so its best to use that and every now and again just clean it. Pete's sanded the plate so it looks brand new and is going to put silicon around the edges.

Danyc0, 2016-12-14:
What's the point in removing it at all if you're just gonna replace it with another brass one?

CloudCoppola, 2016-12-14:
Don't need to replay it now. It was just really dirty so just had to clean it and attach it back on.

Tebazil12, 2016-12-14:
Don't forget part of the issue was that the screws arent the right material and rusted, you'll need to find some decent ones to replace them with (probably stainless steal like tom suggested)

Tebazil12, 2016-12-14:
You might actually want to replace the brass place too btw, brass corrodes in salt water (which is what she'll be sailing in most of the time)

Danyc0, 2016-12-14:
Really there are two acceptable long-term options:
A. Replace it with a plate of a material which won't rust (good stainless steel, fibreglass, carbon fibre)
B. Remove the fiberglass tube used to attach the rudder (use gloves to avoid the fibreglass problem) and replace it with a new, stronger one, maybe carbon fibre (because obviously the forces on the rudder are too strong for fibreglass, because this is the second one of those tubes to crack on us). It'd need to be attached securely to the transom with fibreglass.

TomGC96, 2016-12-14:
I was thinking replacing the tube would be the way to go. Something stronger, and make sure it's on straight too!

closed 2017-06-01 by CloudCoppola