abersailbot/dewi-hardware #72: Diagnose rudder servo problem

kragniz, 2017-05-29:
The rudder servo stopped being powered after test. Figure out what's wrong with it.

Danyc0, 2017-05-30:
By the sounds of what was described, it's either that the gearing has jammed, or the motor has burnt out. I'd say the former is more likely because the photo's show nothing obvious.

TomGC96, 2017-05-30:
The servo has been removed and is broken. It appears that the motor may have burnt out, judging by the smell and the fact that it takes 3 amps by powering it directly from a power supply. A replacement identical servo has been ordered, and will arrive and be fitted soon.

Danyc0, 2017-05-30:
Why order the exact same? Surely that means it might just do the same thing again?

TomGC96, 2017-05-30:
Because now is not really the time to be making big modifications to the way the boat works, considering the competition is so soon. Additionally and maybe more importantly, we have no, and have had no suggestions for any alternatives.

Tebazil12, 2017-05-31:
The last one survived 3+ years so that type is definitely good enough for now.

CloudCoppola, 2017-06-01:
Diagnoses complete.

closed 2017-06-01 by CloudCoppola