abersailbot/dewi-hardware #76: Attach line between antennae and mast

Tebazil12, 2017-06-04:
Rigging can currently wrap around the new deck mounted antennae and get caught, stopping the boat from sailing properly and risks breaking the antennae. 

Best solution may be to glue string to the top of the antennae to tie to the mast. Zip ties were used on the testing weekend but were hard to secure, aren't very reusable and caused the antennae to bend toward the mast as they were pulled too tight (this really isn't good for the antenna).

Danyc0, 2017-06-04:
Really the best solution is to to get an extender and run it up the mast as discussed previously. Improves range (even more) and completely removes the risk of snagging.
It should be a job for 2 / 3 zip ties (which would have to be cut for removing the mast), and completely removes the risk of antenna bend.
It should be really minimal work (just purchase the extension coax and zip tie it to the mast), but to some extent that depends on their availability (delivery times, etc) and cost. @naturesyouth

Tebazil12, 2017-06-04:
That will not happen before this Thursday @Danyc0 , there is already too much work for just Tom to do (everyone is leaving Aber today). The holes for the deck mounts have already been drilled ready for sailbot and we know for sure that like this it works.

Start a new issue for mast mounted antennae for after this competition. We're still not sure on moving the antennae to the mast either way.

Danyc0, 2017-06-04:
I don't quite know what you're envisioning, but what has always been proposed is just connecting a single cable between the deck-mount and the antenna, then zip tying that to the mast for use. It uses the connector which has already been added to the deck.

Tebazil12, 2017-06-04:
Ok, that's different to what I heard before. Are there cables that long with the right connectors that are waterproof that can get here before Thursday? If so we could try it out.

Danyc0, 2017-06-04:
That's always been my plan, I'm glad we're now on the same page. @naturesyouth should know and be able to source one, as he ordered the connectors.

TomGC96, 2017-06-04:
@naturesyouth also paid for them out of his own pocket, and the Bulgin connectors are not cheap so I don't think it's fair to ask him to buy more. The deck antennae increased the range so much I still don't think it's necessary, considering the additional faff. You should also remember there won't be a way to put those antennae flat with the mast, because of the screw connector at its base.

Danyc0, 2017-06-04:
I wouldn't expect him to pay, but either way that's his choice. Remember as well as improving the range, it would also fix the problem that this issue is about, in a way that is a lot less faff (and less permanent) than anything else proposed. What do you mean about the screw connector? I assumed it could just be unscrewed and then folded away with the mast?

TomGC96, 2017-06-04:
The antenna is as seen here: http://bit.ly/2s6AA9V. This blue piece at the bottom attaches to the deck mount in order to seal the connector. As far as I can see, it can't simply be removed.

Danyc0, 2017-06-04:
Wait what? seriously? Why do we bother having a connector at all if it's not removable? haha

naturesyouth, 2017-06-04:
Don't worry, I am here to solve all your antenna problems. First off how much of a problem is the current rigging-antenna setup? on a scale of "would be nice to fix" to "spontaneous unplanned hardware reconfiguration". Also, could I get the diameter and the length of run for the new antenna feed cable?

TomGC96, 2017-06-04:
What I meant, was that we would need it to be like the left image, but with our current antenna, it would look like the one on the right. my.mixtape.moe/kocymo.jpg

Danyc0, 2017-06-04:
what's wrong with the right hand image?

TomGC96, 2017-06-04:
The angle - it's messy, means that the antenna is not straight, and more than anything it puts pressure on the antenna.

naturesyouth, 2017-06-04:
I agree that mounting at that angle is not good, i am gonna suggest fitting a mounting plate on the mast with two bolts and using that to hold the antenna, but i need to know how long the feed cable would need to be.

TomGC96, 2017-06-04:
The problem you have then is that you add a lot of weight to the top of the mast, which will make the boat more unstable. I honestly don't think it's necessary, we were getting massive range out of it in Barmouth. I think we should at least wait until after Sailbot to decide, so we can properly test how good the WiFi and XBee range is with these deck antennae.

naturesyouth, 2017-06-04:
is 200g gonn\a make that much difference? and cat the boat run reliably with the current setup (without any string)

Danyc0, 2017-06-04:
I think at the very least, if it needs a mount, that suddenly becomes a lot more work than is feasible in a couple of days! (on top of what else needs doing).

Re: current setup, It's always a big worry getting the winch cable caught on the antenna, string seems like a bit of a crappy option, especially if it's glued, but I can't really think of anything better for the timescale.

naturesyouth, 2017-06-04:
Ok, so can this issue be fixed by reseating the current deck mounted antenna, at what height does the line interfere on the antenna. And if it's the best solution i can have a extension cable and bracket made up in time for sailbot.

Danyc0, 2017-06-04:
Are you suggesting moving it? If so, that's not feasible in the time, and if not, what do you mean by reseating?
It's not just the fab time, you've got to account for delivery and getting it attached (by someone that isn't you).
Just a quick possibly crap idea, how about sticking a cone over it, so the mainsheet just slides off?

TomGC96, 2017-06-04:
Don't worry, I'll sort something to stop any string or rope from getting stuck on the antennae. It will be better than just some string or cable ties.

naturesyouth, 2017-06-04:
I was planning on fabbing it then attaching it at sailbot, if moving the current bulkhead is a no go how about fitting the second on in a position it doesn't interfere and just capping the first on.

Danyc0, 2017-06-04:
I guess depending on what tom's plans are, there's no harm in you fabbing it all and trying to mount it at sailbot. If it doesn't work/happen/get done in time, just keep what tom has done, if it does work, then great! Also that plan for moving it just sounds horrific.

TomGC96, 2017-06-04:
If you decide to extend the antenna, use the two which are there at the moment, it would be fine to run a cable straight from them as is. They are in pretty much the only position we could see would not interfere with anything else. Preferably, anything new to be done should not interfere with the new paintwork (filling holes in the deck etc.).