abersailbot/dewi-hardware #77: Run Wifi Antenna up mast

Danyc0, 2017-06-04:
See discussion in #76

TomGC96, 2017-10-02:
Bracket has been created, and running the antenna up the mast is now possible, however as proven at Sailbot 2017 this was not necessary. With the current setup on the deck, we experienced WiFi range of up to 200m, and XBee range of well over 350m.

closed 2017-10-02 by TomGC96

colinsauze, 2017-10-02:
Range at Sailbot 2017 may have been helped by an 2nd story balcony used to monitor the boat and receive telemetry. Don't be surprised if range is a lot worse when the ground station is at sea level.

TomGC96, 2017-10-02:
That as it may be, the balcony was only roughly 6 metres higher than the water, and with our distance from the boat being so great, the overall difference in angle between that and us at 2 metres should not be that great.